0xchocolate/flipperzero-esp-flasher #6: ESP 32 Vroom "Enter Bootloader" and "Reset Board" hanging

RatchetHamster, 2023-08-28:
Hi,

Love this project! 

I am running an ESP32 Vroom 32 instead of the Flipper Wifi Dev board. 

When I press, "Enter Bootloader" to try and flash firmware, I get "Entering Bootloader" appear, but then it just hangs (similar error with "Reset Board"). In order to flash firmware, I have to enter bootloader manually using the buttons before I enter "Flash ESP" menu. 

This is fine, but I think it is interfering when trying to switch between Firmware. I have to enter the bootloader using buttons before switching. If I do, it seems to switch fine and tells me to do a manual reset - but even if I use "reset" button or power off and on, it still does not actually switch the firmware, it just sticks to the original firmware.  

Is this something that can be looked into or is it just a limitation of using Vroom that dual boot is off the table?

speedracer1190, 2023-10-28:
I have the same issue. Did you happen to find a fix? I'd love to do the dual boot thing.

b1scoito, 2023-11-13:
Having the same issue here. I've tried setting my ESP32-S3 into bootloader/reflash mode, but the app keeps telling me it's not in reflash mode when I try to flash.

lucidnx, 2024-01-17:
I think you need more wires to reset board, not just RX/TX and GND. ESP32-S need to have EN pin shorted to ground to reset. How is ESPFlasher doing reset and booting to bootloader? @0xchocolate

0xchocolate, 2024-01-17:
@lucidnx The auto-reset into bootloader only works if EN/GPIO0 are connected to RTS/DTR like on the official wifi board (EN on esp32 to RTS pin on flipper, IO0 on esp32 to DTR on flipper)

Without those connections, you should still be able to go into reflash mode manually, if the board has buttons for it.

0xchocolate, 2024-01-17:
Note: RTS is pin 6, DTR is pin 7 here https://docs.flipper.net/development/hardware/wifi-developer-board

jplajpla23, 2024-03-03:
Hi @0xchocolate for what I read about your comments if i connect TX,RX and with some transistors the pin 6 and 7 to my esp32 i can automatically upload firmwares to esp32 from flipper zero withtout touch any button correct?