0xchocolate/flipperzero-esp-flasher #8: Custom partitions.

zap8600, 2023-09-12:
How can I flash custom partitions with this?

0xchocolate, 2023-09-12:
Hey! 👋 I was actually *just* looking at your ai-on-fz repo.

Custom address input is still a TODO. It's not a huge change, but nobody had asked me yet, so I left it on the backburner. Out of curiosity, which addresses do you want to flash?

zap8600, 2023-09-12:
> Hey! 👋 I was actually *just* looking at your ai-on-fz repo.
> 
> Custom address input is still a TODO. It's not a huge change, but nobody had asked me yet, so I left it on the backburner. Out of curiosity, which addresses do you want to flash?

Yeah, I'm not sure that's gonna work due to the Flipper Zero's build system.

Here is my partition table:
```
## Label          type  ST       Offset      Length
nvs,              data, nvs,     0x0000a000, 0x00005000
phy_init,         data, phy,     0x0000f000, 0x00001000
factory,          app,  factory, 0x00010000, 0x000a0000
etc,              0x40, 0x1,     0x000b0000, 0x00070000
linux,            0x40, 0x0,     0x00120000, 0x00460000
rootfs,           0x40, 0x1,     0x00580000, 0x00a70000
```

zap8600, 2023-09-12:
I can try to help implement this if needed.

0xchocolate, 2023-11-26:
This should be next on the roadmap (followed by reading registers). I wasn't sure how the interface would look, but I think I'll add an app option to load a partitions CSV (like the one you posted) and have the browse menu use it.

rnadyrshin, 2024-08-21:
I would also really like to be able to change the addresses of the partitions. This would make your application more versatile.
The implementation method with the CSV file that you mentioned above is not bad. However, I wanted to suggest a couple of other ways the UI could work.
1. When selecting menu items such as "Bootloader", "Part Table", and "NVS", you can display a window for entering an address. The default address is shown, but user can change it.
2. All these partitions are simply binary blocks. They can be flashed one at a time. Flash -> Select address -> Select file -> Go