0xchocolate/flipperzero-esp-flasher #11: updating dual boot marauder

igeometrix, 2023-11-11:
How exactly would I go about updating just the marauder on the dual boot esp flasher. FW B is marauder and I want to update to the newest version plus the version that allows serial pcaps as I only have the wifi dev board. If you have any suggestions that would be amazing. Thanks for all the work you do too btw.. amazing stuff!
Cheers,
Chad

0xchocolate, 2023-11-11:
Flash the marauder bin to the address marked FirmwareB. (Make sure it's not a bootloader bin.)

igeometrix, 2023-11-12:
Before I saw this comment I grabbed the updated marauder dual boot bin (bootloader) from squach and just reflashed the dual boot all together. Is that enough or do I still need to manually go in and reflash the marauder on firmware B as well to get the serial SD files saved.

closed 2024-01-20 by 0xchocolate