0xchocolate/flipperzero-esp-flasher #15: Unable to Flash the Aliexpress esp32 multiboard.

mrDevGit, 2024-01-20:
Hey all. Trying to flash the esp32 of a multiboard with SD card that I bought off aliexpress. I was able to flash the official wifi board with the esp flasher via the flipper but when I attempt it with this board it just tells me to make sure the board is in "reflash mode" and try again. Can anyone help?

- using the esp flasher option to "enter bootloader" after which it says "waiting for download" on the screen
- using firmware from here: https://github.com/justcallmekoko/ESP32Marauder/wiki/update-firmware
- using 3 files not the boot app, as described here: https://github.com/0xchocolate/flipperzero-esp-flasher?tab=readme-ov-file
- This is the multi board with the nrf module, esp32, sd card and sentry safe pins. The selector switch is set to esp32.
- this is the board I have: https://www.aliexpress.com/item/1005006375138591.htmlspm=a2g0o.order_list.order_list_main.4.100118028ZYogo

Should I be using 4 files or 3? 
Should I be using the NON sd_serial firmware?
Should I flash "fast" or "slow"?
Should I flash to firmware "A" or "B"?

I have tried various combinations of the above but can not get it to work.

Thanks

0xchocolate, 2024-01-20:
The board didn't come with update instructions? Have you asked the seller?

Would you be able to take a pic (or screenshot) of the "waiting for download" and maybe a video of the error?

mrDevGit, 2024-01-20:
[image]

[image]

mrDevGit, 2024-01-20:
Does that help?

0xchocolate, 2024-01-20:
Stay on that screen. Unplug the board, and, while holding the boot button on the board, plug it back in. You should see the "waiting for reflash" message again. Then go back and try to reflash.

0xchocolate, 2024-01-20:
If that doesn't work, try restarting the flipper, then plug in the board while holding boot, and go directly to reflash without selecting "Enter bootloader" in the app (just in case that option is interfering and putting the gpio in a bad state).

mrDevGit, 2024-01-20:
I tried all of those options, nothing worked...

mrDevGit, 2024-01-20:
I see that the offical board is the ESP32-S2-Wrover but this one is the  ESP32-S2-Solo. Does that make a difference?

0xchocolate, 2024-01-20:
Still the exact same error?

The revision wouldn't make a difference just for connecting. Try a usb flasher (like fzeeflasher) with the official dev board (S2) target.

mrDevGit, 2024-01-21:
Omg it worked with fzeeflasher! Thanks! I didn't even know that existed! :)

mrDevGit, 2024-01-21:
Though I guess the original issues remains?... that the flash was not possible via the awesome espflasher app?

0xchocolate, 2024-01-23:
Yep, let's leave it open as an issue. Although I don't know how I'll debug this without one of those boards 😄

Rumble74, 2024-01-31:
@mrDevGit what firmware bin file did you end up using for the Esp32 board?  I have the same issue as you and same multi board by your description.  S2-solo

mrDevGit, 2024-02-06:
I used fzeeflasher and selected the SD card option. Worked perfectly. Need to use the Flipper in UART mode.

0xchocolate, 2024-02-06:
Is this still not working in the latest app (v1.5)?

Rumble74, 2024-02-07:
@0xchocolate  I have Flipper firmware 0.98.2,   I have flipperzero-wifi-marauder-f7-api-54.0-0.98.2 from your page and while everything appears to be working as I scan/list/select/deauth/sniff  the files on the flipper zero are present but zero size.  The logs on the other hand appear to be written correctly.

0xchocolate, 2024-02-07:
You need to flash the latest marauder FW (v0.13.7) to your board.

Rumble74, 2024-02-07:
That worked a treat!  Thank you so much!!

0xchocolate, 2024-02-07:
Did you use fzeeflasher or this app to flash?

lol1238, 2024-03-06:
I had the same struggle to get the firmware on the AliExpress Board but I figured out you have to press and hold the boot button und than press the enable or reset button for 2 seconds than everything worked fine for me.

Kip259, 2024-05-23:
> Omg it worked with fzeeflasher! Thanks! I didn't even know that existed! :)

I was have the same issue I went on fzeezy but still have complications 
When I connect the board back after flashing it still asking to connect Wi-Fi module

Parker84757, 2024-05-24:
I am having the same issue as well. What option do you select for board in the fzee flasher site?

Parker84757, 2024-05-24:
Kip259 did you get it to work? I see they have an ali express board option. The other guy said that he used the sd acrd option, but i see mulltiple sd card option devices. Ill let you know if i figure something out either way. I am havingthe same issue as you are. This is prob a decent sized issue. I doubt we are the only ones whao are having these problems. The multi board is actually really legit if we can get it to work. I cannot believe they would go through all the trouble of designing such a board and then offer absolutely zero guidance. I mean not even a small set of instructions.

Kip259, 2024-05-24:
I was thinking the exact same thing. No where to to find a manual or
tutorial

Parker84757, 2024-05-24:
Ok so I got it to work through the esp flash app on the flipper. I used fzeeflasher.com I dont remember if i had to hold the BT button down before plugging the board into the usb cable to my computer, then once it is connected release the BT button. Then click the connect button on the fzee flasher site and it should pop up as com something or other but the name of the device is esp32 S2. then select esp32-s2 as the device. select the latest version, and then select marauder. Then click program. This was the only combination of different devices that came back without any error codes. 
Once you get the board flashed make sure you have the marauder companion app installed. I am running unleashed dev mode and the marauder app from the web flashing site app store was broken and didnt work, so I had to go and reinstall the latest release of unleashed again on my flipper through the unleashed github repo. I selected the web installer that was all the way at the bottom. To get to the web installer i went here  https://github.com/DarkFlippers/unleashed-firmware/releases/tag/unlshd-073

Keep in mind you have to be using chrome web browser to install anything through the web installer. Once I did all of this I was able to go onto my flipper and go to Gpio from the main menu not the apps or browser, and i scrolled all the way down to save to sd card and selected yes. Then go into apps--->gpio--->esp flasher. I then selected quick flash, and selected flipper wifi board/xeon. I am pretty sure thats the version i used to flash to. You will know right away if it is the correct one because if it isnt it will tell you right away that you selected the wrong version. But dont stress bc you can just go back and choose a different one. I am almost positive thats the one that worked for me. It was literally the last one i tried that worked so... Also press the BT button on the gpio board and then click flash and then release the button and it will recognise the board and begin to flash it. Or you can quickly push the BT button after you select flash and it will work. If you dont push the button it will give you some error that i dont remember exactly what it said anymore.
ANYWAY, this was a hassle for me and i have been working on getting this stupid ass thing to work for the last 4 hours and this will hopefully help someone like me who comes looking for advice. CHEERS!

Kip259, 2024-05-25:
Thanks bro I finally got it.. very detailed

Parker84757, 2024-05-26:
> Thanks bro I finally got it.. very detailed
> […](#)

yeah man i got tired oof following others and getting to a point where they got vague or something didnt quite match up... Also i found that i have to go into esp flasher and select firmware b option to get the marauder app to work again each time.

Parker84757, 2024-05-28:
If anyone is interested I contacted the sellers of the multi board and they send me a bunch of .bin files as well as a helpful sheet that tells you exactly how to install the firmware to the device. Maybe one of the devs would want the .bin files to post to their tutorials. I dont know enough yet about this newly aquired hobby yet to implement them into use, although I am trying to learn.

Kip259, 2024-05-28:
Thanks, I'll check it out!