0xchocolate/flipperzero-esp-flasher #16: Invalid chip ID error.

Scottyp1980, 2024-01-20:
I am using the flipper zero wifi dev board esp32, I am getting all the way through the flash and then at the end it tells me In valid chip ID.  Can anyone tell me what I am dealing with?

0xchocolate, 2024-01-20:
Which bin(s) did you flash?

Scottyp1980, 2024-01-21:
esp32_marauder.ino.bootloader_2.bin, esp32_marauder.ino.partitions.bin, and esp32_marauder_v0_13_6_20231214_esp32_lddb.bin  which I renamed firmware in the flipper.

Scottyp1980, 2024-01-21:
when you posed that last question i decided to take a closer look at my bin files and had realized i didnt download the ones for the flipper wifi dev.  I am new to this and have been working on this for three days, none of the other methods seemed to work.  thank you.

Scottyp1980, 2024-01-21:
if i switch to roguemaster will marauder still work?

closed 2024-01-21 by 0xchocolate