0xchocolate/flipperzero-esp-flasher #18: adding evil portal

Scottyp1980, 2024-01-26:
Is there any other route for me to flash evil portal onto my flipper zero wifi dev board without using  flipC.org which is down  at the moment.  Ive succesfully flashed esp32 marauder with the ESPflasher, but i am stuck at this point.  Still new to this.

0xchocolate, 2024-01-27:
What are you trying to do? Just use Marauder's builtin Evil Portal.

flipc.org is irrelevant

Scottyp1980, 2024-01-27:
Thats what I thought, but every time i try to use it built in with the marauder that I flashed using espflasher it doesnt seem to work. Does it make a difference if im running unleashed firmware?  Im going to try it again to see if i missed something.

Scottyp1980, 2024-01-28:
Ah, I was not configuring the SSID properly!  Works great! thank you for taking the time to respond and creating such a great tool.

closed 2024-01-28 by Scottyp1980