0xchocolate/flipperzero-esp-flasher #32: How to flash D1 mini (AZ-Delivery esp8266mod12-f) with Flipper Zero ESP Flasher?

BlkPingu, 2024-10-25:
Since [CH340 drivers](https://www.wch.cn/download/CH341SER_MAC_ZIP.html) for the [ESP8266 D1 mini](https://www.az-delivery.de/en/products/esp32-d1-mini) drivers for Mac are not compatible with MacOS Sonoma I was looking for alternatives to flash my ESP and found out about [ESP flasher](https://github.com/0xchocolate/flipperzero-esp-flasher) for the Flipper Zero.

Has anyone here ever done this?

I can't figure it out from the docs. They are all very concerned about flashing marauder on the wifi boards, but not flashing things like WLED.

Essentially, I need to know:

1. which GPIO Pins I need to connect
2. How to set the D1 mini into DFU mode
3. How to flash the [ESP8266 firmware](https://github.com/Aircoookie/WLED/releases/tag/v0.14.4): meaning apart from the firmware itself, what files do I need to select in the ESP flasher app and where to get them

I saw that there are [files for the ESP32 marauder in this repo](https://github.com/justcallmekoko/ESP32Marauder/wiki/update-firmware#using-spacehuhn-web-updater) thats linked from the the [ESP flasher repo](https://github.com/0xchocolate/flipperzero-esp-flasher) and I suspect they want me to use the rightmost columns files as the header says something about D1 mini: 0x1000, 0x8000, 0xE000 and the WLED firmware instead of the marauder firmware.

I downloaded these files and they all say something about marauder. I don't understand enough here to know if I can use these to flash my WLED firmware as well.

Even if I wanted to try this, I still don't know what pins I need to connect to my D1 mini.

Please advise.

BlkPingu, 2024-11-07:
nobody?

Bald-79, 2024-11-17:
I will not create a separate branch, I will ask here.ESP Flasher seems to support switching tsp32 to firmware mode without pressing buttons. If so, do I need to use additional GPIO pins besides RX and TX? What other exits are needed?

0xchocolate, 2024-11-17:
@Bald-79 To enter the bootloader / download mode automatically, connect EN to flipper pin 6 and GPIO0 to flipper pin 7.

Bald-79, 2025-01-25:
> [@Bald-79](https://github.com/Bald-79)Для автоматического входа в режим загрузчика/загрузки подключите EN к контакту 6 флиппера, а GPIO0 — к контакту 7 флиппера.

It would not be bad if it were pins 15 and 16, because they are on the same connector. In general, it's great that you can change these pins in the application settings.

Bald-79, 2025-01-25:
> [@Bald-79](https://github.com/Bald-79) To enter the bootloader / download mode automatically, connect EN to flipper pin 6 and GPIO0 to flipper pin 7.

It would not be bad if it were pins 15 and 16, because they are on the same connector. In general, it's great that you can change these pins in the application settings.